ros-drivers/openni2_camera #27: Header file linux/usbdevice_fs.h missing on OS X

jmtatsch, 2015-02-12:
When trying to catkin_make openni2_camera on ROS Indigo / OS X 10. 10.10.2, I get the following error:

```
/Users/tatsch/ros_catkin_ws/src/openni2_camera/src/usb_reset.c:40:10: fatal error: 'linux/usbdevice_fs.h' file not found
#include <linux/usbdevice_fs.h>
         ^
```

because that header file is not available under OS X.

As a workaround I replaced the include with

#define USBDEVFS_RESET             _IO('U', 20)

which seems to work. How can this be done properly?

chenbaifan, 2015-05-06:
I got the same error and fix it with your method. I think it's not proper neither, but don't know how to do.

mikeferguson, 2016-01-02:
Fixed by https://github.com/ros-drivers/openni2_camera/pull/30

closed 2016-01-02 by mikeferguson