ros-drivers/openni2_camera #29: Exposing IR emitter properties through ROS?

mklingen, 2015-04-07:
I wonder if it might be possible to expose some of the openni2 emitter properties through ROS in some way? In particular, it would be nice to be able to turn the emitter on and off using a service call which wraps

```
Device.setProperty(XN_MODULE_PROPERTY_EMITTER_STATE,(XnUInt64)(TRUE/FALSE));
```

mikeferguson, 2015-04-07:
Probably possible, pull requests welcome.

mklingen, 2015-04-07:
I was just wondering if it made much sense considering most of the [other properties of this kind](https://github.com/OpenNI/OpenNI2/blob/master/Include/PS1080.h) are set using the [ini file](https://github.com/OpenNI/OpenNI2/blob/master/Config/OpenNI2/Drivers/PS1080.ini) -- but apparently since openni2, they have been made directly accessible.

NikolasE, 2017-11-04:
I tried to implement the feature, but I get an exception if I try to set the state:

```
void OpenNI2Device::setEmitterState(bool enable) throw (OpenNI2Exception)
{
  const openni::Status rc = openni_device_->setProperty(XN_MODULE_PROPERTY_EMITTER_STATE, enable);
  if (rc != openni::STATUS_OK)
    THROW_OPENNI_EXCEPTION("Couldn't set emitter state to %i: \n%s\n", enable, openni::OpenNI::getExtendedError());
}
```

	Failed to set property 1080e007: Device Protocol: Unsupported opcode!
	Device.setProperty(1080e007) failed

Any ideas?