ros-drivers/openni2_camera #33: Unable to start RGB and Depth streams at the same time

r3n33, 2015-05-18:
Running on the ODROID XU3 14.04 I can view
/camera/depth/points but when I enable /camera/rgb/image_raw the point cloud data stops streaming.

There are no errors or warnings present.

r3n33, 2015-05-18:
I should also mention I've tried with the apt pre-compiled binaries as well as compiling OpenNI2 and openni2_camera from source.

This also prevents registered depth points from working.

r3n33, 2015-06-01:
I'm still hunting for what could be the source of the problem. At some point I discovered changing the camera resolution effects the transmission of camera/depth_registered/points and camera/rgb/image_raw.

Example:
https://www.youtube.com/watch?v=xzXhI9OgCUI

r3n33, 2015-06-02:
Looks like a USB bandwidth problem on the ODROID with the USB 2.0 ports. Use the 3.0 port and/or modify openni2_camera to set color_mode to 640x480 @ 15Hz.

closed 2015-06-02 by r3n33

mintar, 2016-05-02:
In case somebody stumbles on this in the future (as I just did): I had the same problem with an ODROID-XU4. Streaming color + depth simultaneously at full 640x480 @ 30 Hz doesn't work on the USB 3.0 ports. Reducing resolution works, or alternatively switching to the **USB 2.0 (!)** hub made it work for me.