ros-drivers/openni2_camera #36: Driver has no re-connection logic

stwirth, 2015-08-05:
When you launch an openni device with this driver, disconnect it from USB and re-connect it, the driver will recognize those events: [OpenNI2DeviceManager::onDeviceDisconnected()](https://github.com/ros-drivers/openni2_camera/blob/0ede5dd527c18d60ca003c4becf29b728bacf97a/src/openni2_device_manager.cpp#L123) and [OpenNI2DeviceManager::onDeviceConnected()](https://github.com/ros-drivers/openni2_camera/blob/0ede5dd527c18d60ca003c4becf29b728bacf97a/src/openni2_device_manager.cpp#L109) are called. However the `OpenNI2Device` handle in `OpenNI2Driver` is not updated. As a consequence, the device will never be re-opened and no images are served.
It would be great if the driver would use the callbacks in `OpenNI2DeviceManager` to reestablish the connection.

130s, 2017-08-17:
This issue is labeled as `enhancement` but IMO it's close to bug (though w/o seeing a spec if such thing exists).

Either way, I'm facing an issue probably caused by what @stwirth described. In our system we noticed that Xtion got disconnected and connected again. Problem is that ROS driver remains disconnected once that happens. This happens only occasionally, likely after a few days of continuous run.

I'm going to look into this. If someone has already a working fix that'll be very appreciated.

130s, 2017-08-25:
Another approach for addressing this issue than what @stwirth mentioned can be to simply exit and  respawn `nodelet` ([this thread](https://answers.ros.org/question/220937/ctrl-c-shutdown-of-nodelet-manager-versus-calling-shutdown-inside-of-nodelet-manager/) might give a hint to do so), which sounds easier to implement. I'll try this way.

130s, 2017-08-25:
I might take back what I said in https://github.com/ros-drivers/openni2_camera/issues/36#issuecomment-324920619 ; By respawning the camera's nodelet the subscribers could stop receiving data after respawning and there might be nothing this driver package can do for that. Meanwhile the way in https://github.com/ros-drivers/openni2_camera/issues/36#issue-99181019 I assume subscribers still can get the data as long as the driver resumes publishing, without any code change on their side. So that is self-contained and more ideal.

130s, 2017-08-31:
Update: We have working patch in our fork that consists of https://github.com/plusone-robotics/openni2_camera/pull/1 and https://github.com/plusone-robotics/openni2_camera/pull/2.

shaun-edwards, 2017-09-01:
The forks @130s mention address re-connection issues along with exposure setting.  If a camera reconnects and auto exposure/white balance and/or exposure were modified, one would like the camera to maintain those settings so that the images are consistent (before and after disconnection).  It turns out, that is easier said than done.  I focused on resetting to a fixed exposure, since that seemed the most straight-forward.  However, the exposure setting functionality was not robust, so I had to add some workarounds to address that as well.

I'm not sure our final solution is the best/right solution, but it achieved our immediate needs.

PR #53 (2017-10-26): Add device re-connection
Fix #36.
Also includes a fix to #51, which would be an issue upon re-connection.
We've been testing this on a system that's running continuously.

Also see discussions in forked repo https://github.com/ros-drivers/openni2_camera/pull/1, https://github.com/ros-drivers/openni2_camera/pull/2

@stwirth I know it's been awhile since your original report #36 but if you could take a look that'll be great.
Of course anyone is welcomed for the review.

Credit goes to @shaun-edwards
closed 2017-10-26 by 130s