ros-drivers/openni2_camera #43: openni2_driver_lib needs a dependency on GetSerial

tfoote, 2016-05-20:
It's failing to build on Xenial 64 due to this: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__openni2_camera__ubuntu_xenial_amd64__binary/22/console

I believe [this line](https://github.com/ros-drivers/openni2_camera/blob/indigo-devel/CMakeLists.txt#L54) needs the additional dependency `${PROJECT_NAME}_generate_messages_cpp`

Surprisingly this is the only platform where it is failing due to this race condition.

jacquelinekay, 2016-05-26:
It would be if we could fix this ASAP (at least before the next Kinetic sync) since currently neither openni2_camera nor freenect_stack are available for Kinetic. @130s

mikeferguson, 2016-05-26:
@jacquelinekay  -- if you can open a PR I can get it merged release -- but I don't have compile/test capabilities at the moment

jacquelinekay, 2016-05-26:
I think #44 will fix this.

jacquelinekay, 2016-06-07:
I think the build is still failing because the fixed version needs a release.

mikeferguson, 2016-06-07:
@jacquelinekay I never saw a "build has returned to normal" on the deb, so I forgot to release -- I see http://build.ros.org/job/Kdev__openni2_camera__ubuntu_xenial_amd64/changes -- releasing now

mikeferguson, 2016-06-07:
https://github.com/ros/rosdistro/pull/11768

closed 2016-06-07 by mikeferguson