ros-drivers/openni2_camera #49: Lunar Release

mikaelarguedas, 2017-06-20:
Howdy openni2_camera maintainers!

It would be really awesome to have a Lunar release of this very widely used package.
Thanks!

mikaelarguedas, 2017-08-05:
Howdy @mikeferguson, any update on this ? it would be really great to have openni2 drivers in Lunar :smile:

130s, 2017-08-17:
@mikeferguson @ros-drivers/openni @wjwwood I can help release stuff if that helps, same as I do for  many other repos. If that sounds good one of you please add me write access here, possible `openni2_launch` and `*-release` repos. Thx.

130s, 2017-08-17:
> write access

Looks like I've already have one for all necessary repos. I'll take care of these.

130s, 2017-08-20:
https://github.com/ros/rosdistro/pull/15795 done

closed 2017-08-20 by 130s

mikaelarguedas, 2017-08-21:
thanks @130s !

130s, 2017-08-28:
Version 0.2.8 also made into Indigo https://github.com/ros/rosdistro/pull/15796
Kinetic https://github.com/ros/rosdistro/pull/15797