ros-drivers/openni2_camera #51: Fails to set RGB exposure on before image streaming

shaun-edwards, 2017-08-29:
I would like to set the RGB exposure level on startup.  I am using ROS Indigo on Ubuntu 14.04 (openni2_camera package built from source).  The exposure level cannot be set unless the camera is actively streaming data, see [Openni2 issue](https://github.com/occipital/OpenNI2/issues/125).  However, even if that issue is resolved, setting exposure on start up will still fail (but in a different way).

The driver either ignores the exposure setting or perhaps after it is fixed returns an error.  Due to the change detection logic, [here](https://github.com/ros-drivers/openni2_camera/blob/indigo-devel/src/openni2_driver.cpp#L306-L307), the dynamic reconfigure will not re-set the exposure unless the exposure is specifically changes.  As a result there value reflected in dynamic reconfigure does not match the value in the low level driver.

I believe this should mostly be fixed in the low level driver code, but the change detection logic should probably be reworked to avoid similar "hidden" bugs.

shaun-edwards, 2017-08-31:
A potential work-around is to set the exposure in the `ColorConnectCb()` method directly on stream start.

shaun-edwards, 2017-08-31:
Setting the exposure in the callback works as long as there is a small delay after the stream is started.

NOTE: Mixing auto white balance and fixed exposure setting rejected by the camera.  If white balance is used at all, the exposure setting is inconsistent.  If using a fixed exposure, it is recommended that auto white balance be disabled on startup.

130s, 2017-10-26:
Fixed via https://github.com/ros-drivers/openni2_camera/pull/53

closed 2017-10-26 by 130s