ros-drivers/openni2_camera #66: disabling hardware depth registration leads to error

130s, 2017-11-04:
[image] **Issue by [bit-pirate](https://github.com/bit-pirate)**
_Monday Sep 23, 2013 at 02:59 GMT_
_Originally opened as https://github.com/ros-drivers/openni2_launch/issues/9_

----

When I deactivated depth registration via rqt_reconfigure, I get the following error:

```
[ERROR] [1379904975.770528069]: Depth image frame id [sensor_3d_depth_optical_frame] doesn't match RGB image frame id [sensor_3d_rgb_optical_frame]
```

Can anyone confirm this? What's different to the old version (openni1)?

130s, 2017-11-04:
[image] **Comment by [bit-pirate](https://github.com/bit-pirate)**
_Monday Sep 23, 2013 at 03:05 GMT_

----

Maybe @piyushk knows what's going wrong.

130s, 2017-11-04:
[image] **Comment by [piyushk](https://github.com/piyushk)**
_Monday Sep 23, 2013 at 16:56 GMT_

----

@bit-pirate I don't have anything other than a kinect at hand at the moment, so I can't test this directly. Can you try and locate which nodelet is generating this error?

130s, 2017-11-04:
[image] **Comment by [bit-pirate](https://github.com/bit-pirate)**
_Tuesday Sep 24, 2013 at 04:15 GMT_

----

It's the registration nodelet (PointCloudXyzrgbNodelet) complaining: 

```
Depth image frame id [sensor_3d_depth_optical_frame] doesn't match RGB image frame id [sensor_3d_rgb_optical_frame]

Location:
/tmp/buildd/ros-hydro-depth-image-proc-1.11.2-0precise-20130909-1058/src/nodelets/point_cloud_xyzrgb.cpp:PointCloudXyzrgbNodelet::imageCb:167
```

Looks to me like the driver is outputting images or point clouds with different reference frames, when not using hardware registration.

130s, 2017-11-04:
[image] **Comment by [piyushk](https://github.com/piyushk)**
_Tuesday Sep 24, 2013 at 15:32 GMT_

----

I suspect the s/w registration pipeline is broken somehow. Once s/w
registered, the depth image should have the RGB image frame id. I'm not
sure how the driver could make this mistake.

130s, 2017-11-04:
[image] **Comment by [mikeferguson](https://github.com/mikeferguson)**
_Wednesday Sep 25, 2013 at 18:35 GMT_

----

@bit-pirate you mention "disable via rqt" -- does software registration work correctly if you start up in that mode?

130s, 2017-11-04:
[image] **Comment by [liborw](https://github.com/liborw)**
_Tuesday Oct 01, 2013 at 08:07 GMT_

----

I have the same problem when started without the hardware registration (not using the dynamic reconfigure).

130s, 2017-11-04:
[image] **Comment by [liborw](https://github.com/liborw)**
_Tuesday Oct 01, 2013 at 08:46 GMT_

----

The problem is, that one must also set `hw_registered_processing` to `false` and `sw_registered_processing` to `true` along with the `depth_registration` to `false`. That way everything works fine.

saikishor, 2018-09-21:
@130s @liborw I am having the same issue, but when setting the depth registration to false via rqt_reconf. If I set it directly when I run my node, there is no such issue. I would like to know whether this issue has been resolved or not?.

130s, 2018-09-22:
Assuming from the last update on this ticket was back in 2013, very few people have had either experienced this issue or been interested in tracking this down. So it'd be great if you @saikishor can look into it deeper, and come up with a workaround to the least (or open a fix via pull request to the best).

mikeferguson, 2024-11-03:
duplicate of #67 (which has slightly more info in it)

closed 2024-11-03 by mikeferguson