ros-drivers/openni2_camera #84: How to use openni2 with kinetic ??

Abduoit, 2018-07-02:
I want to use openni2 with Kinetic as mention here 
https://github.com/ros-gbp/openni2_launch#openni2_launch-kinetic---022-0

Any help is appreciated 

I am getting the following output
`
[ INFO] [1530543321.228722086]: No matching device found.... waiting for devices. Reason: std::__cxx11::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /home/abdu/catkin_ws/src/openni2_camera/openni2_camera/src/openni2_driver.cpp @ 737 : Invalid device number 1, there are 0 devices connected.`

130s, 2018-07-02:
- Closing. Please re-post http://answers.ros.org/, which is the place for this type of questions to get the most response.
- Find a user documentation [docs.ros.org/kinetic/openni2_launch](http://docs.ros.org/kinetic/api/openni2_launch/html/) as [referenced from official wiki](http://wiki.ros.org/openni2_launch#Usage) (I'm not sure how you reached the link you provded...that should not be referenced from anywhere).
- Also for your particular case, make sure the sensor is detected on the OS level, e.g. share system output of `lsusb` command etc.

closed 2018-07-02 by 130s

Abduoit, 2018-07-02:
Thanks for your answer 
I am sure the kinect is detected by `lsusb`.

I provided the link that also answered there.

Does this repo allow using kinectv1 with ros kinetic ?
thank you