ros-drivers/openni2_camera #88: Required arguments id_manufacturer, id_product not set in openni2_tf_prefix.launch

tlind, 2018-09-22:
Commit ff647b83d86c introduced a regression because openni2_tf_prefix.launch was not updated with the arguments `id_manufacturer` and `id_product`, which are since then required by `device.launch.xml`. Instead, openni2.launch was updated correctly.

This means that currently openni2_tf_prefix.launch cannot be used if load_driver is true.

130s, 2018-09-22:
Thanks for the report. Do you think you can open a pull request?
(I'm the one who made the referenced commit but it might take at least a few days for me to get to it).

130s, 2018-09-25:
@tlind Please let me know if you're planning to provide a fix. Otherwise I'll try to squeeze my time into this later this week, as this is bad.

130s, 2018-10-04:
Opened PR https://github.com/ros-drivers/openni2_camera/pull/89

commit 153a6321 (2018-10-20): Merge pull request #89 from 130s/fix88/i

[roswtf plugin] Accept vendor and product IDs (fix #88).
commit 2d2df077 (2018-10-20): [roswtf plugin] Accept vendor and product IDs (fix #88).
closed 2018-10-20 by 130s