ros-drivers/openni2_camera #93: Compatibility with low resolutions

pl-kabaradjian, 2019-01-25:
Hello,

I have been trying to use this package with an in-house device that is a QQQVGA (80x60) depth camera, that can output an IR stream and a Depth stream through a working OpenNI2 proprietary driver. There is no RGB output on this device. 

When trying to naively run the openni2_camera_node it is giving me the famous:

`[ERROR] [1548437429.970623586]: Unsupported IR video mode - Resolution: 640x480@30Hz Format: Gray16`
`[ERROR] [1548437429.971079624]: Unsupported color video mode - Resolution: 640x480@30Hz Format: RGB888`
`[ERROR] [1548437429.971311786]: Unsupported depth video mode - Resolution: 640x480@30Hz Format: Depth 1mm`

Then by taking a deeper look at the code I found that the lowest resolution supported byt the video mode list is 160x120. It is also the case in the dynamic reconfigure.

I tried adding a QQQVGA mode in both the "openni2_driver.cpp" and the reconfiguration cfg but I can't manage to make the openni2_camera_node work. I get rid of the unsupproted mode error but get a segfault instead.
The thing is that the test_wrapper test works fine without any modification and show me the correct resolution for my device.

Output of test_wrapper:

```
IR sensor video modes:
   - Resolution: 80x60@30Hz Format: Gray16

Color sensor video modes:
   - Resolution: 80x60@30Hz Format: RGB888

Depth sensor video modes:
   - Resolution: 80x60@30Hz Format: Depth 1mm
```

Can anyone help on this topic?
It would be nice to make this openni2 wrapper work with even more devices.

closed 2019-04-02 by pl-kabaradjian

tatag4u, 2019-04-15:
Hi I am getting same error.
How did you solve this?

tatag4u, 2019-04-15:
I am getting error as
process[camera_base_link-11]: started with pid [10934]
process[camera_base_link1-12]: started with pid [10967]
process[camera_base_link2-13]: started with pid [10973]
process[camera_base_link3-14]: started with pid [10982]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 004398444747
[Info] [Freenect2Impl] found 1 devices
[ INFO] [1555299351.265756426]: Device "freenect2://0?serial=004398444747" found.
[ INFO] [1555299351.265790728]: Device "freenect2://0?serial=004398444747&depth-size=640x480" found.
[ INFO] [1555299351.265803875]: Device "freenect2://0?serial=004398444747&depth-size=512x424" found.
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns 0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_0_39
libva info: va_openDriver() returns 0
[Info] [VaapiRgbPacketProcessorImpl] driver: Intel i965 driver for Intel(R) Skylake - 1.7.0
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[ERROR] [1555299351.641491726]: Unsupported IR video mode - Resolution: 640x480@30Hz Format: Gray16
[ERROR] [1555299351.641635203]: Unsupported color video mode - Resolution: 640x480@30Hz Format: RGB888