ros-drivers/openni2_camera #100: StringID instead of serial?

MatthijsBurgh, 2020-11-13:
Is there any reason why the StringID of the device is used as serialnumber in the color/ir name instead of serialnumber? This breaks compared to openni, which does uses the real serialnumber.

openni: https://github.com/ros-drivers/openni_camera/blob/indigo-devel/openni_camera/src/nodelets/driver.cpp#L142
openni2: https://github.com/ros-drivers/openni2_camera/blob/noetic-devel/openni2_camera/src/openni2_driver.cpp#L145

Because of this difference the `camera_info_manager` throws a warning: https://github.com/ros-perception/image_common/blob/noetic-devel/camera_info_manager/src/camera_info_manager.cpp#L288

MatthijsBurgh, 2020-11-20:
@reinzor @warp1337 @awesomebytes what is your view on this issue?

awesomebytes, 2020-11-20:
@MatthijsBurgh I haven't used openni_2 in a while, but what you say makes a lot of sense. The only thing I'd be worried by now is backwards compatibility. Maybe there is people that now depend on this behaviour.

Maybe there should be a parameter to actually use the serial number instead of the StringID and by default is false.

reinzor, 2020-11-20:
I agree

MatthijsBurgh, 2020-11-21:
I can do that. I think in the first new ros distro, we should make the serial the default again and the distro after it drop the option to choose.

PR #101 (2021-01-12): Use serialnumber in camera names
Fixes #100
closed 2021-01-12 by 130s