ros-drivers/openni2_camera #103: Release 1.5.0. Branching b/w ROS N, and M or older

130s, 2021-01-12:
Conversation about new release is started hidden at https://github.com/ros-drivers/openni2_camera/pull/101#issuecomment-758609532, so I'm moving it to a ticket.

- For Melodic the [one registered](https://github.com/ros/rosdistro/blob/364bebb6f4c1ce0557fb0ac84dea747d4a30ec01/melodic/distribution.yaml#L7644) as of today is `indigo-devel`. Looking at https://github.com/ros-drivers/openni2_camera/pull/98 the only diff needed b/w ROS M and N is manifest change, which is trivial. But honestly I don't forsee many changes to come in for this pkg so I'm fine to keep separate branches for now (don't touch what's been working principle) unless people think differently.
- Technically this PR is a new feature, let alone bugfix. ROS community is a bit more proactive in adding new features into already-released distro (compared to e.g. `apt`/Ubuntu). I suggest a "happy-medium": Release this into Noetic and Melodic, but not to Kinetic (which is about to EoL in a few months). 
   - If there's convincing voices to support Kinetic release we can as well.
   - Hope making a feat. release into M is ok.

MatthijsBurgh, 2021-01-12:
ABI check succeeds for Kinetic, Melodic and Noetic. So based on that it could be ported back to Kinetic as well.

Based on Kinetic EOL, I agree with you to skip Kinetic.

mikeferguson, 2021-01-12:
Feature releases into older distributions happen all the time - as long as they are API compatible. Some repositories actually enforce ABI compatibility, but they are the minority within the ROS1 ecosystem (mainly those that are low-level and shipping libraries others might actually build on).

I would suggest to rename the noetic-devel branch as "ros1" and use it to release into melodic+noetic. I'd also suggest updating the README for that branch to note that it supports melodic and noetic.

130s, 2021-01-12:
Looks like I beat the discussion to it for Noetic release https://github.com/ros/rosdistro/pull/27961

Branching and adding a note for M and N sounds good.

MatthijsBurgh, 2021-01-26:
@130s when do you plan to release for Melodic?

130s, 2021-01-28:
I think it'd be better, less future burden if we can clean the branching before making a release into ROS1 M as Mike said in https://github.com/ros-drivers/openni2_camera/issues/103#issuecomment-758718607. I'd have to review https://github.com/ros-drivers/openni2_camera/pull/98 again whether a release can be made for both M and N from the same branch.

That said I pushed [ros1](https://github.com/ros-drivers/openni2_camera/tree/ros1) branch made off of `noetic-devel` branch. If anyone can review the branching, and send PRs to ros1 branch if necessary, that'll be great.

MatthijsBurgh, 2021-01-30:
The only changes compared to `0.4.2` (last release of Kinetic and Melodic) are the changes from #101. These are already in the `ros1` branch. I have updated the travis config. See #105. I don't think any other changes are needed. So it can be released into Melodic.

130s, 2021-02-01:
Conditions of satisfaction in order I think of (will be moved to the top post):

# CoS
1. [x] Create a devel branch for ROS1 suggested in https://github.com/ros-drivers/openni2_camera/issues/103#issuecomment-758718607
2. [ ] Fix CI.
1. [ ] Update readme to clarify devel branching as suggested in https://github.com/ros-drivers/openni2_camera/issues/103#issuecomment-758718607
1. [ ] Make releases, have PRs merged on rosdistro, for ROS Noetic, Melodic, Kinetic from the new devel branch.
1. [ ] Remove `noetic-devel` branch (be sure to do this after PRs to rosdistro are merged).

130s, 2021-02-01:
> - Make releases, have PRs merged on rosdistro, for ROS Noetic, Melodic, Kinetic from the new devel branch.

- N https://github.com/ros/rosdistro/pull/28192
- M https://github.com/ros/rosdistro/pull/28193
- K https://github.com/ros/rosdistro/pull/28194

130s, 2021-02-03:
Looks like all PRs https://github.com/ros-drivers/openni2_camera/issues/103#issuecomment-770935498 are merged.

Once synch happens for all distros, latest released installers will be made off of the new `ros1` branch so that `noetic-devel` won't be in use, thus we can delete it at that time IMO.

closed 2022-04-12 by 130s