ros-drivers/openni2_camera #109: openni2_launch with D435?

AlessandroMelino, 2021-03-05:
Hello.

I am facing an issue trying to use the openni2.launch of ROS with a Realsense D435. So when I execute this, appears in terminal:

`[ INFO] [1614940981.625499350]: No matching device found.... waiting for devices. Reason: std::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-noetic-openni2-camera-1.5.1/src/openni2_driver.cpp @ 742 : Invalid device number 1, there are 0 devices connected.`

Thanks in advance.

Best regards.
Alessandro

siddharthcb, 2021-03-19:
same issue with d435i, @AlessandroMelino were you able to solve it?

AlessandroMelino, 2021-03-19:
Hello @siddharthcb.

No, I wasn't able. I have used the realsense2_camera package to launch the camera node (modifying the height and width as desire) and remap the topics to set the same names.

Best regards.

siddharthcb, 2021-03-19:
hello @AlessandroMelino ,

It seems that openni2 is not compatible with realsense cameras.
Not sure what your application was but mine was to convert pointcloud to laserscan.
I could not use openni2 so I just went without it and the problem got solved.

130s, 2022-03-23:
> It seems that openni2 is not compatible with realsense cameras.

This says it.

Realsense is well used in ROS world with the ROS driver Intel makes. See http://wiki.ros.org/RealSense

closed 2022-03-23 by 130s