ros-drivers/openni2_camera #112: Compiling error

jmguerreroh, 2022-03-22:
After compiling using `colcon build --symlink-install` I obtain this error

```
/src/openni2_camera/openni2_camera/include/openni2_camera/openni2_device.h:69:37: error: ISO C++17 does not allow dynamic exception specifications
   69 |                 rclcpp::Node* node) throw (OpenNI2Exception);

```
Any idea?

Juancams, 2022-10-19:
Removing all throw and OpenNI2Exception.

Check https://github.com/Juancams/openni2_camera

130s, 2023-02-22:
C++17 so I assume you're on ROS2 or something newer than what ROS1 supports. AFAIK the packages on this repo are not officially ported to ROS2, it's tracked in https://github.com/ros-drivers/openni2_camera/issues/90

mikeferguson, 2023-02-22:
These exception thingies are gone in the ROS2 branch

closed 2023-03-02 by jmguerreroh