ros-drivers/openni2_camera #124: First ROS2 debian release

130s, 2023-02-23:
1. camera driver wrapper ROS2-fied.
   - [x] Wrapper ported to ros2 https://github.com/ros-drivers/openni2_camera/issues/90
   - [ ] #119
   - [ ] #120
   - [ ] #121
   - Dependencies debian-ified in ROS2.
      - [ ] `rgbd_launch`: ROS2 port is tracked in https://github.com/ros-drivers/rgbd_launch/issues/57. While I'm not sure if openn2_launch still need to depend on it, I'm happy to help porting it.
      - Else?
   - Else?
1. GBP repo
   - [ ] Release team for rgbd, openni2 releases?
   - [ ] Create GBP repo https://github.com/ros-drivers/openni2_camera/issues/90#issuecomment-1441406677
   - [ ] Bloom to ROS `humble` (and `foxy`?).
1. Else?

mikeferguson, 2023-02-23:
I assigned this to myself for now since I'll do the first release into humble/rolling at our current state (it at least gets a mostly working openni2_camera driver out there)

The timeline for the other tickets (and rgbd_launch) is really blocked on the upstream lazy subscriber stuff - and it will only work in Iron and later (those changes aren't going to make it into Humble, since it is a big API change in the ROS2 core)

mikeferguson, 2023-02-26:
PRs required to get team/repo setup:

 * https://github.com/ros2-gbp/ros2-gbp-github-org/pull/210
 * https://github.com/ros/rosdistro/pull/36320

mikeferguson, 2023-02-26:
Also, I've updated the first comment here to reflect that we won't be releasing openni2_launch in the first go-around (there is just too much upstream). This package also really doesn't support foxy anymore because there are too many API changes - so we'll be releasing only to Humble and Rolling (Foxy goes EOL in a little over 2 months anyways)

mikeferguson, 2023-03-02:
Released 2.0.0 to humble and rolling:

 * https://github.com/ros/rosdistro/pull/36369
 * https://github.com/ros/rosdistro/pull/36370

130s, 2023-03-02:
Wondered how you could bloom-release the driver pkg without rgbd_launch. Then found in `CATKIN_IGNORE` in `openni2_launch` pkg. Nice!

mikeferguson, 2023-03-02:
> Wondered how you could bloom-release the driver pkg without rgbd_launch. Then found in `CATKIN_IGNORE` in `openni2_launch` pkg. Nice!

Yeah - that's also how our CI is working. I'm guessing it will be about 3-6 months before we can actually even start to port openni2_launch based on how much upstream stuff has to happen

mikeferguson, 2023-03-02:
- Rolling built fine - http://repo.ros2.org/status_page/ros_rolling_default.html
- Humble binary isn't building right - https://build.ros2.org/view/Hbin_uJ64/job/Hbin_uJ64__openni2_camera__ubuntu_jammy_amd64__binary/ - looks like we need to add a dependency to get PkgConfig? that seems odd that it would show up now....

mikeferguson, 2023-03-03:
#128 should fix the humble build - not sure why our CI didn't detect that - something else must pull in pkgconfig

mikeferguson, 2023-03-03:
New release with humble build fixes, and fixes for use_device_time not working:

- https://github.com/ros/rosdistro/pull/36379
- https://github.com/ros/rosdistro/pull/36380

mikeferguson, 2023-03-06:
Humble build has succeeded:
 * https://build.ros2.org/job/Hbin_uJ64__openni2_camera__ubuntu_jammy_amd64__binary/6/
 * https://build.ros2.org/job/Hbin_ujv8_uJv8__openni2_camera__ubuntu_jammy_arm64__binary/6/

The builds are still failing on RHEL - but that's because libopenni2 doesn't exist there - I think I disable bloom from putting those releases in

mikeferguson, 2023-03-11:
https://github.com/ros2/ros_buildfarm_config/pull/264 - to remove RHEL builds

mikeferguson, 2023-03-13:
Ok - RHEL change is merged - I'm calling this complete

closed 2023-03-13 by mikeferguson