ros-drivers/openni2_camera #130: Unknown CMake command "rosidl_get_typesupport_target".

trichnak, 2023-03-07:
When trying to build package from source for ROS2 Foxy, compiler gives this command when building openni_camera.
[image]

mikeferguson, 2023-03-11:
The ROS2 branch only officially supports Humble and later (as noted in the README on the front page).

I think this change was specifically done to merge the message package back into the main package - you could try running an older branch/commit before we did that update - perhaps try https://github.com/mikeferguson/openni2_camera/commit/e018f7f596a51d348573aaf650dacc0ded2bd242 (which was pre-humble updates)

foxydevelby, 2023-03-31:
> The ROS2 branch only officially supports Humble and later (as noted in the README on the front page).
> 
> I think this change was specifically done to merge the message package back into the main package - you could try running an older branch/commit before we did that update - perhaps try [mikeferguson@e018f7f](https://github.com/mikeferguson/openni2_camera/commit/e018f7f596a51d348573aaf650dacc0ded2bd242) (which was pre-humble updates)

Hello @mikeferguson, thank you for creating this pkg.  I was able to run this and get pointcloud data, but when I displayed this in rviz2, no pointcloud is seen. I tried changing the fixed frame to openni_rgb_optical_frame then set the pointcloud QoS to Best Effort but no to avail.....Is this a TF related issue ? is the old branch has it? or I need to create one?

mikeferguson, 2024-01-25:
Foxy is now EOL - so we can safely close this ticket (all supported ROS 2 distros are supported by the ros2 branch)

closed 2024-01-25 by mikeferguson