ros-drivers/openni2_camera #139: build error

anfederman, 2024-05-29:
: note: declared here
   74 | CameraPublisher create_camera_publisher(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/ros2_ws/src/openni2_camera/openni2_camera/src/openni2_driver.cpp:198:58: error: too many arguments to function ‘image_transport::CameraPublisher image_transport::create_camera_publisher(rclcpp::Node*, const string&, rmw_qos_profile_t)’
  198 |     pub_depth_ = image_transport::create_camera_publisher(this, "depth/image", custom_qos, pub_options);
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ubuntu/ros2_ws/src/openni2_camera/openni2_camera/include/openni2_camera/openni2_driver.h:39,
                 from /home/ubuntu/ros2_ws/src/openni2_camera/openni2_camera/src/openni2_driver.cpp:32:
/opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:74:17: note: declared here
   74 | CameraPublisher create_camera_publisher(
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/openni2_camera_lib.dir/build.make:76: CMakeFiles/openni2_camera_lib.dir/src/openni2_driver.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:656: CMakeFiles/openni2_camera_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< openni2_camera [21.9s, exited with code 2]

Summary: 22 packages finished [41.1s]
  1 package failed: openni2_camera
  1 package had stderr output: openni2_camer.....

Hi Mike - Working on Mockturlebot project with Camp - trying to  old Asus working.

mikeferguson, 2024-05-29:
This is almost certainly a versioning issue - make sure you're using the right branch of openni2_camera - there is an "iron" branch which should work on Humble/Iron, the "ros2" branch requires Jazzy or Rolling (because it implements lazy subscribers, through these function calls using added parameters). The branch notes are in the README

anfederman, 2024-05-29:
Thanks! Hope you checkout HBRC tonight.

closed 2024-05-30 by mikeferguson

anfederman, 2024-05-30:
Thanks works great:

***@***.***:~$ ros2 topic list
/camera/depth/camera_info
/camera/depth/image
/camera/depth/image/compressed
/camera/depth/image/compressedDepth
/camera/depth/image/theora
/camera/depth/image_raw
/camera/depth/image_raw/compressed
/camera/depth/image_raw/compressedDepth
/camera/depth/image_raw/theora
/camera/ir/camera_info
/camera/ir/image
/camera/ir/image/compressed
/camera/ir/image/compressedDepth
/camera/ir/image/theora
/camera/projector/camera_info
/camera/rgb/camera_info
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/theora
/parameter_events
/rosout